efabless/cace #54: Non-schematic projects still try to generate a netlist

jwise, 2024-04-21:
It would be nice to be able to use CACE for all sort of circuit characterization, not just magic-flow IC characterization.  For instance, there are PCB designs that may have no layout, and testbench designs that have no schematic and only a netlist.

Currently, CACE will try to process a schematic into a netlist (and will fail if it does not), even if a netlist already exists.  The desired behavior is that if a netlist exists, and a schematic does not, it should simply attempt to use the netlist that is already there.

mole99, 2024-04-22:
I think a workaround for now could be to place an empty schematic with an older timestamp than the netlist. Then CACE should not try to generate a new netlist.

RTimothyEdwards, 2024-04-22:
@mole99 :  We did that workaround yesterday;  yes, it definitely works to touch an empty schematic and then touch the netlist.  But it's an awful hack.  The fix is just a few conditionals in the cace_regenerate.py code.

jwise, 2024-04-22:
Yeah, I think my big problem with that is that if the timestamps get out of sync in some way (i.e., I check things into git, and then check them out, or something?), then I would end up with my netlist getting blown away.

mole99, 2024-04-24:
Hi @jwise, could you check if #56 fixes this for you? Thanks!