efabless/cace #72: Startup behavior

RTimothyEdwards, 2024-06-03:
I previously had two behaviors in CACE that appear to have become lost:

(1) When `cace-gui` is run, if there is no matching project name or no matching .txt or .yml file, then the GUI still pops up, and waits for the user to select a characterization file to load.

(2) Before `cace-gui` brings up the GUI window, all output text is collected in a "messages" variable.  If the GUI comes up successfully, then "messages" is dumped into the console window pane.  If something happens and `cace-gui` dies, then "messages" is dumped to the terminal as the last thing that happens before the program exits (requiring that exceptions be caught and handled gracefully).

Right now, if I start `cace-gui` and (I think) either of the above items happens, then `cace-gui` returns to the terminal prompt with absolutely no indication that anything happened.

mole99, 2024-06-04:
Sorry about that, I've changed a few things recently and the order in which the functions are called is just wrong!

closed 2024-06-04 by mole99

mole99, 2024-06-04:
Fixed by #75

PR #75 (2024-06-04): Add progress bar and render Markdown + pass `--batch` to ngspice
This PR uses [Rich](https://github.com/Textualize/rich)  to add a progress bar and render the Markdown of the summary.

Also reorders some functions in the GUI to fix #72.

Passes `--batch` to ngspice to exit simulations on error.