efabless/cace #73: After exiting cace the terminal has no visible prompt and no cursor but command entry still functions

azwefabless, 2024-06-03:


azwefabless, 2024-06-04:
Clarification. When exiting from Cace-gui operations by using the close window function in top RHS there is no returned cursor in the linux terminal. Command entry works and terminal responses are visible but the cursor is invisible and text entered is not visible.

mole99, 2024-06-04:
Thanks for the more detailed description! I have experienced this too, but only when I killed cace-gui while running simulations.

But I am not able to reproduce your case. If I just run `cace-gui` and close it via the close button at the top, I still have terminal output.

I think I now know what the issue might be. When cace-gui is started it redirects stdout to the console in the gui. Maybe in your case after closing cace-gui the previous state is not restored. Could you try running `cace-gui --terminal` (which does not redirect stdout) and see if the issue persists?

azwefabless, 2024-06-04:
To create the issue I need to open cace-gui and run a simulation. Any simulation.Same exact issue happens even when I do cace-gui --terminal  . After exit the terminal has a functional but invisible cursor.