efabless/cace #74: Very slow Cace simulation on ulpcomp2 input offset example

azwefabless, 2024-06-03:
cell is ~/cheetah_v3_analog/dependencies/sky130_icrg_ip__ulpcomp/
Run time is exceptionally long for me but about 100x faster for Tim. Suspect an IO setting difference that Tim has set but I have not. 
The simulation with the speed problem is input offset voltage

azwefabless, 2024-06-04:
Yes, Correct repo and correct file. There are 2 sims that are dying in what appears to be similar manner. Both output enormous amounts of Reference value data to the GUI. Tim runs the same config file and it finishes in minutes for him. Mine is still going after 20 minutes with 8 railed CPU cores. Sequential might be a good workaround but is not a fix. I will try it today.

mole99, 2024-06-04:
Thanks! I just ran it on my side (fresh pull) and here it seems to work well:

[image]

I believe it could be a tooling issue. I am using the Nix to set up my tools, it uses the stable version of ngspice.
There's no installation documentation on that yet, but it's basically the same as in OL2: https://openlane2.readthedocs.io/en/latest/getting_started/common/nix_installation/index.html

Which tool versions do you use?

azwefabless, 2024-06-04:
I'm having the issue after swithing to a prereease version of the PDK that Tim requested. Also note that there are a lot of convergence related comments in the terminal. I suspect either a models issue or an environment setting variable around convergence or ngspice setup that is missing from the default config information.