efabless/cace #108: Remove green checkmark from docs Markdown

hpretl, 2024-08-26:
There is a green checkmark in the summary section of the result Markdown. This special characters fails in some environments, and it is not really needed in my view. Please just stick with a "Pass"/"Fail" verdict.

hpretl, 2024-08-26:
@mole99 @RTimothyEdwards Please take a look if you agree.

mole99, 2024-08-26:
Hi Harald, can you tell me under which environment it fails?

I personally like the green checkmark ✅ and the red cross ❌ to see at a glance if a parameter failed or everything was successful.

OpenLane 2 already makes heavy use of Unicode and ANSI escape codes and any modern terminal emulator should have support for this. (You too now Tim 😉)

hpretl, 2024-08-26:
It fails when compiling the `md` with Quarto/Pandoc into Latex to compile to a PDF :-) Sounds complicated, but I use it to do this: https://iic-jku.github.io/analog-circuit-design/#t-ota-simulation-versus-pvt

hpretl, 2024-08-26:
The alternative is that I do post-processing of the Md to remove these characters, as I do currently with the figures paths (need to tweak the paths). Do you have any idea how to do this with a `sed`?

mole99, 2024-08-26:
@RTimothyEdwards Yes, we could try to use alternative characters. How about 🮱 or ✓ and 🮽 or ⨯ ?

hpretl, 2024-08-26:
Whatever these characters are, they do not even render correctly in my browser :-( 

[image]

mole99, 2024-08-26:
It should look like: 

[image]

That is really strange, they seem to be basic Unicode symbols. Does your terminal render them correctly?

hpretl, 2024-08-26:
I changed the Latex compiler to Xelatex, and now I can render w/o fail, although it still does not show this caracter. I use now a script to remove all unicode characters :-)

hpretl, 2024-08-26:
Not sure about terminal, but also my mail program can not render it (all macOS).

mole99, 2024-08-26:
Your browser and mail program probably use the same font which does not support these characters. Are you using the default macOS font? If so, then I'm a bit surprised that they're not supported, but ✅ and ❌ both work?

RTimothyEdwards, 2024-08-26:
@hpretl :  Just be aware that beyond the check-mark and X-mark, unicode symbols are being used for basic engineering text like square root (as in 1/sqrt(Hz)), squared, micro, degree, and a handful of others.  Some of those are in the ISO font encoding and some in the standard symbol font, but I don't know how universally they display.  But just watch out for the possibility that wiping all unicode out of the output will screw up a lot of units.

hpretl, 2024-08-26:
@RTimothyEdwards Good point! Maybe I need to selectively remove Unicode characters then. But so far, I avoid Unicode wherever I can, and use things like uV and rtHz. :-)