efabless/cace #111: Generated plots should show parameter limits

hpretl, 2024-08-28:
Often it is really useful to have parameter limits in generated data plots. This allows an easy check if and with how much margin curves are inside or outside limits. Since all the information is in the YAML datasheet, this should be possible.

hpretl, 2024-08-28:
@mole99 @RTimothyEdwards What do you think about this feature?

mole99, 2024-08-28:
That sounds like a good idea! A dashed [`axhline`](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.axhline.html) could be used for xyplots to mark the limits.

hpretl, 2024-08-28:
Sounds good! Looking forward to the next version(s) will all the great enhancements we have been discussing. :-)

mole99, 2024-08-28:
Me too, now they just have to get implemented 😉

mole99, 2024-08-28:
It seems the implementation isn't that simple ^^

|                    |                    |                    |
|--------------------|--------------------|--------------------|
|  [image]                  | [image]                  |         [image]           | 

If the limits are drawn unconditionally, then for some plots the axis is scaled in such a way that you can no longer see the values properly.

The solution I've come up with is to take the yrange (distance between min/max yvalue) and check if the limit that should be drawn is within that range. With a bit of a margin, that seems to work well:

|                    |                    |                    |
|--------------------|--------------------|--------------------|
|      [image]            |         [image]           |               [image]     |

Only the first plot shows the limit, as it is close enough to the data points. If any of the lines would intersect the limit, it would always be shown as it lies inside the yrange.

mole99, 2024-08-28:
Alright, then I would suggest the following setting for the `plot` entries in the datasheet:

- `limits`
  - `false` - do not show limits
  - `true` - always show limits
  - `auto` - show limits if they are in the yrange

With the default value `auto`.

Example:

```yaml
    plot:
      gain_vs_temp:
        type: xyplot
        xaxis: temperature
        yaxis: a0
        limits: true
      ugf_vs_corner:
        type: xyplot
        xaxis: corner
        yaxis: ugf
        limits: false
      pm_vs_vdd:
        type: xyplot
        xaxis: vdd
        yaxis: pm
        limits: auto
```

Would this work for you?

hpretl, 2024-08-28:
Yes, that is a good proposal!

mole99, 2024-08-29:
Let me know if #114 works for you and I'll merge it.

hpretl, 2024-08-29:
What is the easiest way of testing it? I guess I need to pull it and install it with `pip install .`, right?

mole99, 2024-08-29:
Yes exactly! There's also a target in the Makefile: `make install`

mole99, 2024-08-29:
Thanks for the feedback! Going to merge it and close this issue.

PR #114 (2024-08-29): Implement plot limits
- Implement plot limits
  - Plots have an optional argument `limits`, which can be `true`, `false` or `auto`

Example entry:

```yaml
    plot:
      gain_vs_temp:
        type: xyplot
        xaxis: temperature
        yaxis: a0
        limits: true
      ugf_vs_corner:
        type: xyplot
        xaxis: corner
        yaxis: ugf
        limits: false
      pm_vs_vdd:
        type: xyplot
        xaxis: vdd
        yaxis: pm
        limits: auto
```

Fixes #111
closed 2024-08-29 by mole99