efabless/cace #121: CACE should be able to parse expressions

areda0, 2024-10-03:
A feature that's too useful is that CACE be able to parse expressions, reference conditions, substitutes for variables, and evaluates the final numeric value. The use case is for example to define the spec target/limit as a function in one of the conditions so that the spec target varies with varying conditions.. referring to the snapshot, CACE should be able to evaluate "Vdvdd  - 0.2" and compare the result against that value. 
[image]

mole99, 2024-10-03:
We can take a look at the [expression-engine](https://openlane2.readthedocs.io/en/latest/reference/configuration.html#expression-engine) in OpenLane 2 and see which parts could apply to CACE.

There are quite a lot of design decisions to be made. In which sections of the datasheet can expressions be used?
Can expressions refer to all keys in the datasheet or only to conditions? If the former, how do we specify the "path" to the key? Or can only keys in the same dictionary be referenced? How do expressions behave together with includes?

This is a very useful feature, but requires careful planning for the implementation.

areda0, 2024-10-03:
@mole99 
- Expressions can be used anywhere a numeric value is defined >> having expressions in Parameter specs and conditions values is useful.   
- I think it can refer to conditions and measured parameters, both would have useful use cases . But a parameter is a simulation result and would require to feedback the parameter value after a specific sim has completed. 

@RTimothyEdwards  
what do you think Tim?