efabless/cace #150: Performance of initial runset/parameter creation

MrHighVoltage, 2025-02-27:
Hi all,

I'm currently investigating using cace for huge parameter-sweeps with MC iterations and I recognized, that the initial creation of the netlists etc... takes incredibly long. For example, I'm currently running 3870 simulations and the time to setup the simulation, before actually starting any simulations is around 30 minutes. This problem, of course, increases with bigger parameter sweeps.

The overhead is quite significant and hurts even more on machines with high core counts but low single thread performance.

As a solution, I'm thinking of two possibilities:

- The straightforward would be to create the files required for the simulation "on the fly"/"lazy", so when starting the actual sim. The behavior could be controllable by a switch (e.g.: --lazy-create-simfiles).
- The second one would be to parallelize the creation of the files, maybe also controllable by a switch (--create-simfile-jobs N)

Do you see any chance to implement such a feature?

Thanks in advance and best regards,
Georg

mole99, 2025-02-27:
Hi Georg,

yes indeed, the creation of the testbenches is currently very slow, I've noticed that too.
I think both of your suggestions are a good idea + debugging whether the code could be sped up.

Unfortunately I won't get to it soon, as things are currently very busy...

Br,
Leo

RTimothyEdwards, 2025-02-28:
@MrHighVoltage :  A simulation starting time of 30 minutes sounds very much like how ngspice _used_ to behave for the sky130 process.  This was due to the heavily binned device models, and an issue in ngspice where it was reading all devices for all corners before selecting the corner to be used in the simulation.  That issue was fixed in ngspice a number of major revisions ago.  So I would like to check what is your version of ngspice, and are you using it in conjunction with a .spiceinit file that sets the correct options for the PDK (@mole99 :  Will CACE automatically copy and use the PDK .spiceinit file?).

MrHighVoltage, 2025-02-28:
Hi Timothy, hi Leo,

the actual simulation part was running super fast. Just to give you a number, on our 128-core ARM Server, the initial creation of 650 Parameters took around 6 minutes, the actual simulation was done in two minutes.

Anyways, I'm pretty much on bleeding edge (ngspice 44.2) and using the IHP-PDK. It is really just the creation of the netlists for ngspice, that takes a long time.

@mole99 Ah yes, the usual problem. Time :) Thanks anyways for the replies! I'd keep it open or now, maybe we can label the issue as enhancement.